Seeed-Studio/OPL_Kicad_Library #1: TQFP100 used in Atmega2560 doesn't allow autorouting

autotel, 2017-06-28:
I don't understand well why, this package doesn't allow routing. There seems to be some size or placement of elements that irritate the clearances of the routes, effectively making routing or auto routing impossible.
I replaced the seedOPL version with the sparkfun version, and it works. Maybe the problem could be figured out by comparing the two footprints.

CourageGuo, 2017-06-29:
Hi Autotel
Here is our engineer reply.

Please change the routing grid same as the grid size.
[image]
[image]

autotel, 2017-07-03:
thanks! I solved this problem on my own. But this may help future designers.