Seeed-Studio/OPL_Kicad_Library #3: SW4-SMD-5.2X5.2X1.5MM pad placement and size is different than specification

autotel, 2017-06-28:
The datasheet recommended footprint is different from the footprint in seed OPL. I have made a version according to this recommendation, that I can share.

CourageGuo, 2017-06-29:
Hi Autotel

Could you tell me more information about footprint? Our engineer said it is the same from datasheet.
[image]
[image]

autotel, 2017-07-03:
I am not an expert, so you should trust better the engineer. But this kind of things happen all the time, that is why I wanted to help pointing it out. 

The guide to the footprints that I used is just at the right of the image you attached. Please carify me if I did it wrong.

[image]

CourageGuo, 2017-07-04:
Hi Autotel

Yes it is right.  Please follow the datasheet.

closed 2019-01-22 by SeeedFusion

mfaccin, 2019-08-02:
Hi, I would like to know if @SeeedFusion corrected it.

autotel, 2019-08-02:
I don't actually know; probably. Anyways it is always good to measure them
in comparison to the data sheet. Can you confirm yourself?