Seeed-Studio/OPL_Kicad_Library #8: KiCad Format error

sethhillbrand, 2018-09-04:
The following files are incorrectly formatted for KiCad.  They specify zero-width lines for graphical items.  This is not allowed in the KiCad format.  It will output incorrect gerber files that differ from the screen display and may be dis-allowed in future versions.

Please update to specify explicit line width for graphical items.

```
SeeedOPL-Connector-2016.pretty/3P-L14.2_W11.0_H9.0MM.kicad_mod
SeeedOPL-Connector-2016.pretty/BAT2P-13.5-CR1220.kicad_mod
SeeedOPL-Connector-2016.pretty/DCJ3P-14.2X11.0X9.0MM.kicad_mod
SeeedOPL-Connector-2016.pretty/H10-2.54.kicad_mod
SeeedOPL-Connector-2016.pretty/H2-2.54.kicad_mod
SeeedOPL-Connector-2016.pretty/H2X3-2.54.kicad_mod
SeeedOPL-Connector-2016.pretty/H3-2.54-90D-MALE.kicad_mod
SeeedOPL-Connector-2016.pretty/H3-2.54.kicad_mod
SeeedOPL-Connector-2016.pretty/H4-2.54.kicad_mod
SeeedOPL-Connector-2016.pretty/H6-2.54-90D-MALE.kicad_mod
SeeedOPL-Connector-2016.pretty/H6-2.54.kicad_mod
SeeedOPL-Connector-2016.pretty/MIC2-D9.7XH4.5MM.kicad_mod
SeeedOPL-Diode-2016.pretty/LED4-SMD-5050.kicad_mod
SeeedOPL-IC-2016.pretty/DFN8G-0.5-2X2MM.kicad_mod
SeeedOPL-IC-2016.pretty/DFN8G-0.5-3X3MM.kicad_mod
SeeedOPL-IC-2016.pretty/MSOP8-0.65-3X3MM.kicad_mod
SeeedOPL-IC-2016.pretty/MSOP8G-0.65-3X3MM.kicad_mod
SeeedOPL-IC-2016.pretty/QFN32G-0.5-5X5MM.kicad_mod
SeeedOPL-IC-2016.pretty/QFN44G-0.5-7X7MM.kicad_mod
SeeedOPL-IC-2016.pretty/SOIC20G-1.27-16X11MM.kicad_mod
SeeedOPL-IC-2016.pretty/SOIC8-1.27-5X4MM.kicad_mod
SeeedOPL-IC-2016.pretty/SOP20-1.27-12.7X7.65MM.kicad_mod
SeeedOPL-IC-2016.pretty/SOT-23-6.kicad_mod
SeeedOPL-IC-2016.pretty/SSOP16-0.635-5X4MM.kicad_mod
SeeedOPL-IC-2016.pretty/SSOP20-0.65-7.6X5.2MM.kicad_mod
SeeedOPL-IC-2016.pretty/SSOP24-0.65-8.9X5.3MM.kicad_mod
SeeedOPL-IC-2016.pretty/SSOP28-0.65-10.2X5.2MM.kicad_mod
SeeedOPL-IC-2016.pretty/TQFP100-0.5-14X14MM.kicad_mod
SeeedOPL-IC-2016.pretty/TQFP32-0.8-7X7MM.kicad_mod
SeeedOPL-IC-2016.pretty/TQFP44-0.8-9.6X9.6MM.kicad_mod
SeeedOPL-IC-2016.pretty/TSSOP14-0.65-5X4.4MM.kicad_mod
SeeedOPL-IC-2016.pretty/TSSOP20-0.65-6.5X4.4MM.kicad_mod
SeeedOPL-IC-2016.pretty/TSSOP24-0.65-7.8X4.4MM.kicad_mod
SeeedOPL-LED-2016.pretty/LED2-2.54-D5.0MM.kicad_mod
SeeedOPL-Sensor-2016.pretty/DFN10-0.5-3.0X3.0MM.kicad_mod
SeeedOPL-Sensor-2016.pretty/LGA14-0.8-5.0X3.0MM.kicad_mod
SeeedOPL-Sensor-2016.pretty/QFN24G-0.4-3X3MM.kicad_mod
SeeedOPL-Sensor-2016.pretty/SOP16-1.27-10.0X5.0MM.kicad_mod
```

Huangw0107, 2018-11-27:
Updated，thank you

SeeedFusion, 2019-01-22:
Updated, thank you!

closed 2019-01-22 by SeeedFusion